IFL-CAMP/easy_handeye #87: End Effector frame location

nilic993, 2021-06-14:
Hello, I've achieved an accurate calibration of my UR3 arm using your package, but when I'm publishing TF afterwards the end effector of the arm is misplaced. Here's how the setup looks, both live and in ROS:

[image]
[image]

The frame "tool0_controller" is correctly recognized as the end effector frame, but as you can see in ROS the held object (marker) is located at "wrist_3_link" and not the position of the gripper (effector).

As far as I can tell the "tool0_controller" frame is assigned as the end effector frame in all relevant places during setup, so I'm having trouble figuring out why the gripper and object are moved "backwards" in ROS. Here's the calibration .yaml that's published for TF:

[ur3.txt](https://github.com/IFL-CAMP/easy_handeye/files/6646801/ur3.txt)

marcoesposito1988, 2021-06-20:
Hi @nilic993,

the simplest explanation is that the size of the marker does not agree to that passed to the launch file. If the physical dimension is smaller than the configured, one, the marker will appear farther away from the camera, and this would match your error.

Can you please check with a ruler the dimension of the marker, and compare it to the one you pass as argument?

nilic993, 2021-06-21:
I've checked the marker size (it's physically 5cm and that's how it's passed in the launch file) but apparently it was only a matter of calibration quality:

[image]

Moving the arm (and marker) closer to the camera during calibration improved the end effector position significantly.

closed 2021-06-21 by nilic993

marcoesposito1988, 2021-06-21:
Happy to hear that.. wish you the best of luck with your project!