IFL-CAMP/easy_handeye #88: Unable to see calibration GUI

Yitian-Xian, 2021-06-16:
Hi all, I am using this package on melodic + ROS + Ubuntu 18.04 + UR5 robot.

As suggested by others, I have three launch file: the first one is responsible for opening the kinect: `roslaunch kinect2_bridge kinect2_bridge.launch`.

The second one opens the aruco marker package and ur5 driver(also the moveit planning things): `roslaunch easy_handeye ur5_kinectv2_calibration.launch`. the file can be seen as follows:

`<?xml version="1.0"?>
<launch>
    <arg name="namespace_prefix" default="ur5_kinect_handeyecalibration" />
 
  <arg name="robot_ip" doc="The IP address of the UR5 robot" />
		
   <arg name="marker_size" doc="Size of the ArUco marker used, in meters" default="0.05" />
 <arg name="marker_id" doc="The ID of the ArUco marker used" default="582" />
 
    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/kinect2/hd/camera_info" />
        <remap from="/image" to="/kinect2/hd/image_color_rect" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="kinect2_rgb_optical_frame"/>
        <param name="camera_frame"       value="kinect2_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>
   
      
    <include file="$(find ur_modern_driver)/launch/ur5_bringup.launch">
        <arg name="limited" value="true" />
        <arg name="robot_ip" value="169.254.203.110" />
    </include>
 
    <include file="$(find ur5_moveit_config)/launch/ur5_moveit_planning_execution.launch">
        <arg name="limited" value="true" />
    </include> 
    
</launch>`

The third one opens the calibration program and the rviz GUI: `roslaunch easy_handeye easy.launch`. It looks like:

`<?xml version="1.0"?>
<launch>

 
<arg name="namespace_prefix" default="ur5_kinect_handeyecalibration" />
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="false" />
 
        <arg name="tracking_base_frame" value="kinect2_rgb_optical_frame" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="base_link" />
        <arg name="robot_effector_frame" value="wrist_3_link" />
 
        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>
 
</launch>`

However, the result is very strange. One program just shows empty and the other one seems good. I have attached the photo below.

[image]

Can anyone help me with this problem? Thank you!!!

Yitian-Xian, 2021-06-16:
By the way, this was nothing wrong in the command terminal when I ran the third command, so I do not have any idea how to debug...

marcoesposito1988, 2021-06-20:
Hi @Yitian-Xian,

this is weird, I don't manage to replicate it.

Can you please try out [easy_handeye_demo](https://github.com/marcoesposito1988/easy_handeye_demo), to exclude that the problem is related to the hardware configuration or to your launch files?

OwenKZ, 2021-08-17:
I face the same question. And I am using kinova gen3-lite and realsense d435i. I have no idea why the rqt_easy_handeye program is empty. 
Can you please tell me how to fix it if you have already solved this problem?

YoanDev, 2022-04-01:
I solve this issue.
It's very strange, I went in the directory of scripts in rqt_easy_handeye and I launch the python file rqt_easy_handeye.
The python file doesn't work but, after, you can launch rviz and the GUI will be not empty.

marcoesposito1988, 2022-09-24:
This seems to be due to stale rqt configuration. Please reopen if the issue happens again after using the proposed solution (`rosrun rqt_gui rqt_gui --clear-config`)

closed 2022-09-24 by marcoesposito1988