IFL-CAMP/easy_handeye #90: TypeError: not all arguments converted during string formatting

ravijo, 2021-06-16:
Thank you very much for creating this package.

Based on the [documentation](https://github.com/IFL-CAMP/easy_handeye#publishing), I am using `publish.launch` as shown below:
```
<include file="$(find easy_handeye)/launch/publish.launch">
    <arg name="eye_on_hand" value="false"/>
    <arg name="namespace_prefix" value="my_eob_calib"/>
</include>
```
However, it shows the following error upon running it:
```
Traceback (most recent call last):
  File "/home/ravi/ros_ws/src/easy_handeye/easy_handeye/scripts/publish.py", line 15, in <module>
    rospy.loginfo("filename", filename)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/core.py", line 190, in loginfo
    _base_logger(msg, args, kwargs, level='info')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/core.py", line 183, in _base_logger
    logfunc(msg, *args, **kwargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1174, in info
    self._log(INFO, msg, args, **kwargs)
  File "/usr/lib/python2.7/logging/__init__.py", line 1293, in _log
    self.handle(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 1303, in handle
    self.callHandlers(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 1343, in callHandlers
    hdlr.handle(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 766, in handle
    self.emit(record)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rosgraph/roslogging.py", line 245, in emit
    record_message = _defaultFormatter.format(record)
  File "/usr/lib/python2.7/logging/__init__.py", line 465, in format
    record.message = record.getMessage()
  File "/usr/lib/python2.7/logging/__init__.py", line 329, in getMessage
    msg = msg % self.args
TypeError: not all arguments converted during string formatting
[camera_eye_on_base/handeye_publisher_dell_20316_2678869232382820748-4] process has died [pid 20376, exit code 1, cmd /home/ravi/ros_ws/src/easy_handeye/easy_handeye/scripts/publish.py __name:=handeye_publisher_dell_20316_2678869232382820748 __log:=/home/ravi/.ros/log/c45c30e8-ce78-11eb-84a7-7478271dc0ee/camera_eye_on_base-handeye_publisher_dell_20316_2678869232382820748-4.log].
log file: /home/ravi/.ros/log/c45c30e8-ce78-11eb-84a7-7478271dc0ee/camera_eye_on_base-handeye_publisher_dell_20316_2678869232382820748-4*.log
```

I noticed that because `filename` is empty, the following line produces the error. https://github.com/IFL-CAMP/easy_handeye/blob/0c2e44eecd2ad89a2f752434020de1115d79704e/easy_handeye/scripts/publish.py#L15

Workaround
------------------

Move the line `rospy.loginfo("filename", filename)` from `Line#15` to `else` block 

https://github.com/IFL-CAMP/easy_handeye/blob/0c2e44eecd2ad89a2f752434020de1115d79704e/easy_handeye/scripts/publish.py#L17-L20

nickhward, 2021-06-17:
Hello, I had the same error as you and your workaround did fix it but it resulted in another error. I was currious if you got the same?

`process[my_eob_calib_eye_on_hand/handeye_publisher_nicholas_Alienware_Aurora_R7_21129_5205827542548380165-1]: started with pid [21146]
Traceback (most recent call last):
  File "/home/nicholas/catkin_ws/src/easy_handeye/easy_handeye/scripts/publish.py", line 20, in <module>
    calib = HandeyeCalibration.from_filename(filename)
  File "/home/nicholas/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_calibration.py", line 265, in from_filename
    with open(filename) as calib_file:
IOError: [Errno 2] No such file or directory: 'mycalibration'
[my_eob_calib_eye_on_hand/handeye_publisher_nicholas_Alienware_Aurora_R7_21129_5205827542548380165-1] process has died [pid 21146, exit code 1, cmd /home/nicholas/catkin_ws/src/easy_handeye/easy_handeye/scripts/publish.py __name:=handeye_publisher_nicholas_Alienware_Aurora_R7_21129_5205827542548380165 __log:=/home/nicholas/.ros/log/36d8cf80-cef8-11eb-bacd-d89ef3936c09/my_eob_calib_eye_on_hand-handeye_publisher_nicholas_Alienware_Aurora_R7_21129_5205827542548380165-1.log].
log file: /home/nicholas/.ros/log/36d8cf80-cef8-11eb-bacd-d89ef3936c09/my_eob_calib_eye_on_hand-handeye_publisher_nicholas_Alienware_Aurora_R7_21129_5205827542548380165-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

nickhward, 2021-06-17:
Hello, I had the same error as you and your workaround did fix it but it resulted in another error. I was currious if you got the same?

[image]

ravijo, 2021-06-17:
@nickhward 

Where is your calibration file located? In general, it is inside `$HOME/.ros/easy_handeye/` directory. Once you find the file, you can tell this information to `publish.launch` file using the following command:

```
<include file="$(find easy_handeye)/launch/publish.launch">
    <arg name="eye_on_hand" value="false"/>
    <arg name="namespace_prefix" value="my_eob_calib"/>
    <arg name="calibration_file" value="/home/nicholas/.ros/easy_handeye/mycalibration.yaml"/>
</include>
```

nickhward, 2021-06-17:
@ravijo 

I just realized that the calibration script wasn't showing up when I run my calibration launch file. 

When I run it the frames look correct on rviz but there is no calibration script: 

[image]

ravijo, 2021-06-17:
@nickhward 

I guess you are using older version of OpenCV (version 3 or less). You need to update your OpenCV by using the command below:

```
pip2 install opencv-python==4.2.0.32
``` 

Next, you should check your MoveIt configuration. I don't know why but from your screenshot, it seems that MoveIt is not configured properly.

Then you should try to perform the calibration.

nickhward, 2021-06-17:
Unfortunately I still get the same error with opencv. Are there more steps that need to be taken after the pip install?

ravijo, 2021-06-17:
@nickhward 

You need to make sure that `cv2.CALIB_HAND_EYE_TSAI` is available in your OpenCV. Please see below for more information:

```
ravi@dell:~/ros_ws$ source devel/setup.bash
ravi@dell:~/ros_ws$ ipython
Python 2.7.17 (default, Feb 27 2021, 15:10:58) 
Type "copyright", "credits" or "license" for more information.

IPython 5.5.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import cv2

In [2]: cv2.__version__
Out[2]: '4.2.0'

In [3]: cv2.CALIB_HAND_EYE_TSAI
Out[3]: 0L
```

nickhward, 2021-06-17:
Okay I see now my problem is that even after I have done the pip install of the opencv 4.2 version I still see it as version 3.3-dev after entering in cv2.__version__

nickhward, 2021-06-17:
@ravijo 
I know I'm asking a lot of questions and I really appreciate you helping me. But do you know how to change the version? Nothing on the internet has been helping me achieve this goal.

nickhward, 2021-06-17:
@ravijo 

So I ended up fixing my issue. In case anyone runs into the same issue as me. I had to find the cv2.so file from the newer version of cv2 that I installed. Then copy that into the directory of the old version of cv2 which was in /opt/ros/kinetic/lib//python2.7/dist-packages/

marcoesposito1988, 2021-06-20:
Thanks again @ravijo for providing the error report and the assistance!

@nickhward sorry for the inconvenience, I did not test that change with Kinetic. 

Can you move your arm with MoveIt? If yes, it may use a different move group name than `manipulator`, which is a common convention. You can find out the group name by starting MoveIt for your arm as usual, then looking into the UI of the `MotionPlanning` RViz plugin, `Planning` tab, `Query group` box, `Planning Group` combo box. If it is not `manipulator`, you can pass it to `calibrate.launch` as the `move_group` parameter.

Please open a new issue if you encounter further problems, since this is going to be closed with a commit for ravijo's problem.

commit ab6f1c1c (2021-06-20): Fix crash while publishing

Fixes #90, credits to @ravijo
commit 94ba0dc7 (2021-06-20): Fix crash while publishing

Fixes #90, credits to @ravijo
commit f6b7b0a9 (2021-06-20): Fix crash while publishing

Fixes #90, credits to @ravijo
closed 2021-06-20 by marcoesposito1988