IFL-CAMP/easy_handeye #94: Calibration_server Error

freesolo92, 2021-08-06:
Hey guys,

tried to setup the eayse_handeye Calibrator with a Realsense D435 Camera and an UR3 Robot.
I used the eye_on_base setup, where the camera has a fix point and one aruco marker is on the flange of the Robot.

I tried several times doing calibration with seperate launch files and with one launch file include all others, but no one works for me. 
When I start my own launch file (it also opens the calibration.launch), only rviz will be open automatically and no rqt_gui from easy_handeye.
When I look it the terminal I get the following error due to the calibration_server:

[easy_handeye_eye_on_base/easy_handeye_calibration_server-3] process has died [pid 20376, exit code 1, cmd /home/jetson/test_ws/src/easy_handeye/easy_handeye/scripts/calibrate.py __name:=easy_handeye_calibration_server __log:=/home/jetson/.ros/log/f66e0672-f693-11eb-9707-48b02d3d0639/easy_handeye_eye_on_base-easy_handeye_calibration_server-3.log].
log file: /home/jetson/.ros/log/f66e0672-f693-11eb-9707-48b02d3d0639/easy_handeye_eye_on_base-easy_handeye_calibration_server-3*.log

Does anyone know how to deal with this error?

Thanks for helping me

Regards, Marcel

closed 2021-09-19 by freesolo92

Olimoyo, 2022-05-27:
Did you end up finding a fix?

marcoesposito1988, 2022-10-16:
If you are using melodic you can either install a newer version of python-opencv as [here](https://github.com/IFL-CAMP/easy_handeye/issues/78#issuecomment-767714349) or downgrade easy_handeye as [here](https://github.com/IFL-CAMP/easy_handeye/issues/104#issuecomment-12570843080

hello128world, 2024-03-29:
I had a same error, did you konw how to fix it?