IFL-CAMP/easy_handeye #96: What should I do to use with Xarm6

Aki92-Xavier, 2021-09-26:
Hi,I am trying to use easy_handeye with realsense D435 and Xarm6.
Do you have any suggestion.

marcoesposito1988, 2021-11-21:
As suggested by the documentation, you should use the official ROS Moveit config provided by the robot producer to connect to the robot. For the realsense camera, you can also find examples in the easy_handeye docs, for example https://github.com/IFL-CAMP/easy_handeye/blob/master/docs/example_launch/panda_realsense_eyeonbase.launch

closed 2022-06-17 by marcoesposito1988

Akumar201, 2022-09-28:
Hello, I would like to know in the above example how can I use other camera than realsense ? I want to use simple camera such that data transmission can be efficient even after using long wire.