IFL-CAMP/easy_handeye #97: some issues when i using easy_handeye to calibrate realsense_camera.

AtariaLee, 2022-03-14:
when i using easy_handeye to calibrate my camera .
when i run `roslaunch handeye_fanuc_lee rs_fanuc_eyehand.launch`,I can start my planning program and camera.
 But the Easy Handeye program does not start. 
The error message is below:
`[easy_handeye_calibration_server_robot-12] process has died [pid 20022, exit code 1, cmd /home/lee/catkin_ws/src/easy_handeye/easy_handeye/scripts/robot.py __name:=easy_handeye_calibration_server_robot __log:=/home/lee/.ros/log/f5a07cc2-a389-11ec-bf60-000c29e5bbe1/easy_handeye_calibration_server_robot-12.log].
log file: /home/lee/.ros/log/f5a07cc2-a389-11ec-bf60-000c29e5bbe1/easy_handeye_calibration_server_robot-12*.log
`

`[easy_handeye_eye_on_hand/easy_handeye_calibration_server-13] process has died [pid 20028, exit code 1, cmd /home/lee/catkin_ws/src/easy_handeye/easy_handeye/scripts/calibrate.py __name:=easy_handeye_calibration_server __log:=/home/lee/.ros/log/f5a07cc2-a389-11ec-bf60-000c29e5bbe1/easy_handeye_eye_on_hand-easy_handeye_calibration_server-13.log].
log file: /home/lee/.ros/log/f5a07cc2-a389-11ec-bf60-000c29e5bbe1/easy_handeye_eye_on_hand-easy_handeye_calibration_server-13*.log
`
it almost take me a week to find how to solve this problem,but i didnt find any answer.
So ,does someone konw how to fix it ? I would be very appreciated!!!

AtariaLee, 2022-03-17:
i solved this porblem. This is "moveit_controller_manager.launch.xml"  that caused the error.

closed 2022-03-17 by AtariaLee

feifanrensheng, 2022-04-18:
> i solved this porblem. This is "moveit_controller_manager.launch.xml" that caused the error.

I have the same error. How this porblem was solved?

AtariaLee, 2022-04-20:
@feifanrensheng 
you can try change `ros_controller` at  moveit_controller_manager.launch.xml  to `controllers` , it worked for me ,but i dont konw why it worked.