IFL-CAMP/easy_handeye #98: can‘t open Hand-eye Calibration automatic movement

Minpig, 2022-05-04:
Hand-eye Calibration automatic movement

the other two interfaces can be opened，only the interface about “Hand-eye Calibration automatic movement” cannot be opened，why？my environment is：ubuntu16.04，python2.7，opencv2.4.9. Do I want to upgrade to Ubuntu 18.0 04 or upgrade opencv version?
 
[image]

Minpig, 2022-05-04:
I tried to install opencv4.2.0 with the source code,.After installation, I found that  “Hand-eye Calibration automatic movement”  still didn't appear

Minpig, 2022-05-05:
iIhad solved.

closed 2022-05-05 by Minpig

Akumar201, 2022-09-28:
Hello, I am using the same arm and getting the same error, can you please help me solving this ?
[image]
<launch>
    <arg name="namespace_prefix" default="xarm_realsense_handeyecalibration" />
    <arg name="robot_ip" doc="The IP address of the xarm robot" />
    <arg name="robot_dof" doc="The degree of freedom of the xarm" />
    
    <arg name="marker_size" doc="Size of the ArUco marker used, in meters" default="0.1" />
    <arg name="marker_id" doc="The ID of the ArUco marker used" default="105"/>

    
    <include file="$(find realsense2_camera)/launch/rs_camera.launch" />

    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/camera/color/camera_info" />
        <remap from="/image" to="/camera/color/image_raw" />
        <param name="image_is_rectified" value="true"/> 
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="camera_color_optical_frame"/>
        <param name="camera_frame"       value="camera_color_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
    </node>

    
    <include if="$(eval arg('robot_dof')==7)" file="$(find xarm7_moveit_config)/launch/realMove_exec.launch">
        <arg name="robot_ip" value="$(arg robot_ip)" />
        <arg name="show_rviz" value="false" />
    </include>
    <include if="$(eval arg('robot_dof')==6)" file="$(find xarm6_moveit_config)/launch/realMove_exec.launch">
        <arg name="robot_ip" value="$(arg robot_ip)" />
        <arg name="show_rviz" value="false" />
    </include>
    <include if="$(eval arg('robot_dof')==5)" file="$(find xarm5_moveit_config)/launch/realMove_exec.launch">
        <arg name="robot_ip" value="$(arg robot_ip)" />
        <arg name="show_rviz" value="false" />
    </include>

    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
                
        <arg name="eye_on_hand" value="true" />
        <arg name="tracking_base_frame" value="camera_color_optical_frame" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="link_base" />
        <arg name="robot_effector_frame" value="link_eef" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.1" />
        <arg name="robot_acceleration_scaling" value="0.1" />
        <arg name="move_group" value="xarm$(arg robot_dof)" />
    </include>
    
    <node name="recognition_view" pkg="image_view" type="image_view" args="image:=/aruco_tracker/result" />
</launch>

Minpig, 2022-10-09:
I didn't encounter this problem. I'm sorry I can't help you

------------------&nbsp;原始邮件&nbsp;------------------
发件人: ***@***.***&gt;; 
发送时间: 2022年9月29日(星期四) 凌晨0:25
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; "State ***@***.***&gt;; 
主题: Re: [IFL-CAMP/easy_handeye] can‘t open Hand-eye Calibration automatic movement (Issue #98)

 
Hello, I am using the same arm and getting the same error, can you please help me solving this ?

Akumar201, 2022-10-11:
Thank you for your response, I have done the calibration, can you please suggest me how can I cross check that my calibration results are good. Looking forward to hearing back from you. Thank you in advance.

marcoesposito1988, 2022-10-16:
You can use the tf visualization to inspect the quality of the calibration. If it is correct, while moving the robot around you should see that the position of the tf frames in the world is plausible. Otherwise they will fly around (as it happens during the calibration process)