IFL-CAMP/easy_handeye #99: Hand Eye calibration with gripper or without gripper

goyalvarun02, 2022-05-24:
Hi All, 

I am using xArm7 with xArm Gripper and Real sense camera with eye-on-base configuration. I have query regarding calibration that hand-eye calibration should be done with gripper mounted Aruco code on front of xArm Gripper  or without gripper .

I tried both scenario but the coordinates I receive from real sense camera are not exact center of object.
Sometime it is just front of object and can pickup but not confidently or sometime it missed the object with little value.

What should I do ?
Is there any way to validate the hand-eye calibration is correct?
How can I modify the coordinates to pickup the object from its center?
How to handle the xArm Gripper open/close if one side of finger is just above the object ?

marcoesposito1988, 2022-06-17:
Hi @goyalvarun02,

you can mount the marker on the gripper or directly on the flange of the robot, as long as it can rotate about all axes (best if it can move in all 6 degrees of freedom).

What you have to pay attention to is what to give as `effector_frame` parameter to `easy_handeye`. The marker should be solidly attached to this frame, or the calibration will be bogus.
The other usual suspects are the incorrect size of the AR marker, the calibration of the camera (this one shouldn't be an issue for a realsense, since they are pre-calibrated), outliers in the samples (marker too far away and tracking stopped, lag between marker tracking and robot position, etc)

Please check the [troubleshooting](https://github.com/IFL-CAMP/easy_handeye/blob/master/docs/troubleshooting.md) and the [calibration evaluator](https://github.com/IFL-CAMP/easy_handeye/blob/master/rqt_easy_handeye/src/rqt_easy_handeye/rqt_calibration_evaluator.py)

goyalvarun02, 2022-08-02:
Hi @marcoesposito1988 

Thanks fir the reply . I have multiple queries which are given below . I will really appreciate it if you can provide the solution or pointer for all :-) 
Scenario 1 : If I mount the marker on flange, then it will calibrate the flange position with camera . What will happen after I attached gripper and offset for that gripper along z axis 172 cm as per TCP coordinate system ? 

Scenario 2: If I mount the marker on edge of gripper (xArm), I faced issue as below : 

a) Can not calibrate from current position ? How can I find the starting position is accurate for the calibration ? otherwise I have to move xArm random position and try it again whether now its fine or not.

b) If after hit and trial I found one position is fine it will be set as Home. But after 2 -3 samples later I observed "Pre-emption" error . May I know the root cause of this pre-emption occurrence?

c) After calibration is done successfully (I did not face any error), while pickup and place object, I realized the coordinate of object was return not correct? As it was some plus/minus x/y/z ?  What might be issue for this ?

d) Is there any way or tool , I can identify the calibration is correct or not after generating the file easy_handeye_on_base.xml?

e) Is there any guideline to keeping the EE and camera .  As I can see xArm has limit upto 700mm along x/y/z axis, when I move the robot to starting/observing position at x=500mm and Z=500mm  (which I think suitable) it observe the  object. Thereafter it has very limited scope in X/Z axis for pickup the object. What should I do for this.

f) While using Move IT , I observed object was just kept vertical down to gripper , but move It return fraction of 15% , and sometime also I feel that the trajectory from this position is quite easy and can pick by robot but then I found fraction is <90% . What might be the issue , and what needs to be done for this ? How can I check trajectory and why move it is returning fraction (computeCartesianPath) which is  <90% for simple locations?
 
Thanks

marcoesposito1988, 2022-09-24:
Hi @goyalvarun02,

you can place the marker anywhere on the robot, as long as it can move in all six degrees of freedom during the calibration procedure (translation and rotation in all axes). The position of the marker with respect to the robot links is anyway unknown, and the calibration algorithm is designed to factor it out.

I don't know what you are referring to with the "preemption" error. Can you share more details?

Please check the [troubleshooting](https://github.com/IFL-CAMP/easy_handeye/blob/master/docs/troubleshooting.md) for identifying issues with the accuracy of the result. You can use the calibration evaluator to measure the accuracy as shown [here](https://github.com/marcoesposito1988/easy_handeye_demo/blob/master/launch/check_calibration.launch#L59).

I cannot give support for MoveIt in general. You could try asking on their forum or on https://answers.ros.org

closed 2024-02-10 by marcoesposito1988

maheshr24, 2024-12-02:
Hey if u tried in simulation what should be the camera position 
 quat="0.707 0 0 0.707 " I was trying this orientation on link5