IFL-CAMP/easy_handeye #101: Information on working ROS Noetic Setup with PyQt5

Olimoyo, 2022-05-30:
Hi,

I was looking for some advice from someone who was able to get this package working with ROS Noetic.

The main issue I'm having is  indirectly related to this package and has to do with `PyQt5`. I'm currently using the following versions:
```
PyQt5                         5.15.6
PyQt5-Qt5                     5.15.2
PyQt5-sip                     12.10.1
```
When I run the following command `rosrun rqt_gui rqt_gui`, I get the following error:
```
ValueError: PyCapsule_GetPointer called with incorrect name
```

If someone was able to get this to work with ROS Noetic, what version or setup of PyQt5 did you use? 

Thanks!
Oliver

dongyi-ur, 2022-06-13:
I have the same problem.
And for my case, it mainly because the conda conflicts with the calibration package.
I creat a new workspace and run it in my computer's env (not Virtual Environment) successfully.

Olimoyo, 2022-06-13:
Would it be possible for you to type `pip list` and to see what version's of PyQT worked for you on your computer's env?

dongyi-ur, 2022-06-14:
dongyi@dongyi-ur:~$ pip show PyQt5
Name: PyQt5
Version: 5.14.1
Summary: None
Home-page: None
Author: None
Author-email: None
License: None
Location: /usr/lib/python3/dist-packages
Requires: 
Required-by:

dongyi-ur, 2022-06-14:
[image]

dongyi-ur, 2022-06-14:
[image]

Olimoyo, 2022-06-14:
Thank you! I'll give it a try in the next few days.

marcoesposito1988, 2022-09-24:
This seems to be a conflict with conda. Please reopen the issue if this occurs with a clean installation of 20.04/noetic.

closed 2022-09-24 by marcoesposito1988