IFL-CAMP/easy_handeye #106: Support of easy handeye for ROS2 is available?

goyalvarun02, 2022-09-20:
Hi ,

May I know if github of easy handeye for ROS 2 is available  Please share the path .

or if not available May I know when can expect the ros2 of easy_handeye.

Thanks

marcoesposito1988, 2022-09-24:
Hi @goyalvarun02,

a ROS2 version is under (slow) development [here](https://github.com/marcoesposito1988/easy_handeye2), but there is no timeline yet for its completion.

For the time being you can try using it with the ROS1 bridge.

marcoesposito1988, 2023-08-03:
The ROS2 version should now be usable (up to the automatic robot movement, which may be feasible from Iron upwards and will be the next item on the roadmap), however it was not tested on real hardware yet