IFL-CAMP/easy_handeye #109: How to make sure that camera on base calibration is correct

Akumar201, 2022-09-29:
Hello, I am doing camera on base by setting <arg name="eye_on_hand" value="false" /> as false in calibration launch file. How can I make sure that the calibration I am doing is correct ? 

translation: 
  x: 0.4705739455942296
  y: 1.0399431884657737
  z: 0.6388135790952124
rotation: 
  x: -0.11958812174791901
  y: -0.7127805157974599
  z: 0.6887959517916398
  w: 0.056592881458372546

[image]

marcoesposito1988, 2022-10-16:
You can use the tf visualization to inspect the quality of the calibration. If it is correct, while moving the robot around you should see that the position of the tf frames in the world is plausible. Otherwise they will fly around (as it happens during the calibration process).

Alternatively you can use the `rqt_calibration_evaluator` script, as shown [here](https://github.com/marcoesposito1988/easy_handeye_demo/blob/master/launch/check_calibration.launch#L59)

Akumar201, 2022-10-17:
Thank you for your response. I will do that, In addition to this I resolved the issue by manually measuring the distance from base frame to camera in x, y and z for initial testing, I will use the above mentioned methods for accuracy of calibration. 
Thank you again

closed 2022-10-17 by Akumar201