IFL-CAMP/easy_handeye #113: Use camera on base callibration result.

Akumar201, 2022-10-11:
Greetings, 
I have calibrated the camera using camera on base. I would like to know how can I use this data to pick object. I know I have to make changes in https://github.com/xArm-Developer/xarm_ros/blob/master/xarm_vision/d435i_xarm_setup/launch/publish_handeye_tf.launch
which is for camera on hand. In addition to this do I have to change the https://github.com/xArm-Developer/xarm_ros/tree/master/xarm_vision/d435i_xarm_setup/config file also.

marcoesposito1988, 2022-10-16:
Hi @Akumar201,
I am not affiliated with that package, but it does look like it is using easy_handeye. So you can insert the result of the calibration into the `.yaml` file, or provide the `calibration_file` launch argument to point to the one resulting from your calibration (which by default is in `~/.ros/easy_handeye`)

Akumar201, 2022-10-17:
Thank you for your response, just need to make sure that I am doing it correctly, so I just need to change the .yaml file or direct the launch file to use the desired one.

Akumar201, 2022-11-04:
This issue can be resolved using the [easy_handeye/easy_handye/launch](https://github.com/IFL-CAMP/easy_handeye/tree/master/easy_handeye/launch)  launch file, you need to set the    to false in-case you want to do eye on hand and take care of namespace_prefix.  namespace_prefix will help the program to find out what whether your result will be eob or eoh. 
For example 

```
   <!-- start easy_handeye -->
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="my_eob_calib" />
        <!-- eye on hand: true -->        
        <arg name="eye_on_hand" value="false" />
        <arg name="tracking_base_frame" value="camera_color_optical_frame" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="link_base" />
        <arg name="robot_effector_frame" value="link_eef" />

        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.1" />
        <arg name="robot_acceleration_scaling" value="0.1" />
        <arg name="move_group" value="xarm$(arg robot_dof)" />
    </include>
    <!-- in case rqt_easy_handeye image viewer fails to open: -->
    <node name="recognition_view" pkg="image_view" type="image_view" args="image:=/aruco_tracker/result" />
</launch>
```
 I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2022-11-04 by Akumar201