IFL-CAMP/easy_handeye #117: How to measure the calibration error in a qutantitative method?

wangqingyu985, 2022-11-04:
I'm wondering: Is there a specific way to measure the error of the hand-eye calibration in a quantitative way?
Many thanks!!

Akumar201, 2022-11-04:
Hi @wangqingyu985  I would like to suggest you to measure x, y and z from the (0,0,0) frame of the robot (base or end-effector) to the camera using measuring tape, regarding the orientation vector you can estimate by changing coordinate frame of reference using right hand rule. You will get an approximation initially that if you are doing it right or not. For accuracy purpose you can use fiduciary markers. Please let me know if you need any further assistance with this issue.

wangqingyu985, 2022-11-05:
Thanks for you reply @Akumar201 
I am using Franka robot and RealSense camera with eye-on-base method. And I am sure that the calebration result is right, cause I can realize the visual recognition and grasping function. I would like to measure the calibration error for accuracy purpose, so how to do this with fiduciary markers?

marcoesposito1988, 2023-06-21:
Hi @wangqingyu985,

you can try the `check_calibration` functionality from `easy_handeye_demo`, as here https://github.com/marcoesposito1988/easy_handeye_demo/blob/master/launch/check_calibration.launch#L59