IFL-CAMP/easy_handeye #123: The GUI viewer is empty when I launch the calibration (thus cannot collect samples).

DanielTakeshi, 2023-01-13:
Current system:

- Ubuntu 18.04
- ROS Melodic

I am launching:

```
roslaunch bimanual_ros eye_on_base_calibrate.launch
```

The `bimanual_ros` is a package that I'm using which is loosely based on `franka_ros`. I can provide further details if it is needed (it's a private package someone else gave to me to handle bimanual manipulation). Here is the launch file `eye_on_base_calibrate.launch`:

<details>
<summary>
Here is the file.
</summary>

```
<?xml version="1.0" ?>
<launch>
    <arg name="eye_on_hand" default="false" />

    <!-- these parameters depend on the robot; you can copy these for the Franka Panda -->
    <arg name="move_group" default="panda_arm" />
    <arg name="move_group_namespace" value="/easy_handeye_eye_on_hand" />
    <arg name="robot_base_frame" value="panda_link0"/>
    <arg name="robot_effector_frame" value="panda_hand"/>

    <!-- these parameters depend on the tracking system; these are only valid for this simulator, you'll have to find out yours -->
    <arg name="tracking_base_frame" value="rgb_camera_link"/> <!-- azure -->
    <arg name="tracking_marker_frame" value="aruco_tag"/>

    <arg name="markerId"        default="582"/>
    <arg name="markerSize"      default="0.05"/>  <!-- in m, 0.034 -->
    <arg name="corner_refinement" default="LINES" /> <!-- NONE, HARRIS, LINES, SUBPIX -->

    <!-- azure -->
    <!-- <include file="$(find azure_kinect_ros_driver)/launch/driver.launch" />  -->
    <include file="$(find azure_kinect_ros_driver)/launch/driver.launch" >
        <arg name="overwrite_robot_description" value="False" />
        <arg name="fps" value="30" />
        <arg name="color_resolution" value="720P" />
        <arg name="depth_mode" value="NFOV_UNBINNED" />
    </include>

    <group ns="rgb" > <!-- azure -->
      <node pkg="image_proc" type="image_proc" name="image_proc"/>
    </group>

    <node pkg="aruco_ros" type="single" name="aruco_single">
        <remap from="/camera_info" to="/rgb/camera_info" /> <!-- azure -->
        <remap from="/image" to="/rgb/image_rect_color" /> <!-- azure -->
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg markerSize)"/>
        <param name="marker_id"          value="$(arg markerId)"/>
        <param name="reference_frame"    value="rgb_camera_link"/>  
        <param name="camera_frame"       value="$(arg tracking_base_frame)"/>
        <param name="marker_frame"       value="aruco_tag" />
        <param name="corner_refinement"  value="$(arg corner_refinement)" />
    </node>

    <!-- start easy_handeye calibration procedure -->
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="start_rviz" value="false" />
        <arg name="publish_dummy" value="false" />

        <arg name="move_group" value="$(arg move_group)" />
        <arg name="move_group_namespace" value="$(arg move_group_namespace)" />

        <arg name="eye_on_hand" value="$(arg eye_on_hand)" />
        <arg name="tracking_base_frame" value="$(arg tracking_base_frame)" />
        <arg name="tracking_marker_frame" value="$(arg tracking_marker_frame)" />
        <arg name="robot_base_frame" value="$(arg robot_base_frame)" />
        <arg name="robot_effector_frame" value="$(arg robot_effector_frame)" />

        <arg name="freehand_robot_movement" value="true" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>

    <node name="image_view" pkg="image_view" type="image_view" respawn="false" output="screen">
       <remap from="image" to="/aruco_single/result" />
    </node>

</launch>
```
</details>

When I run this, it correctly shows an image of the top-down scene with detection of the AR Marker. 

But, the GUI is empty!

Here is a screenshot of what I see. 

[image]

Has anyone experienced this before? Confusingly, someone else who was sharing my machine is able to run the same command on a different user account and get the GUI to appear. I have also tried re-cloning this repository and redoing `catkin build`. 

Strangely I can move around this box that’s by the blank GUI?!? Right-clicking will show check-boxes for “Form” and “Minimized dock widgets”.

Here is a video link which shows what I am seeing (it's about 770 MB) in case it is more clear:

https://drive.google.com/file/d/1EgpPZwgD59RCMzzCYaiJdZPF_g9o3v8W/view?usp=sharing

Let me know if it would help to know more information.

DanielTakeshi, 2023-01-14:
To clarify, I can run `rqt` by itself (just after `roscore`, so no robot involved) and it shows a GUI. So it seems like for that command the GUI appears.
[image]

marcoesposito1988, 2023-03-22:
Hi @DanielTakeshi,

it's impossible to tell what is going wrong without seeing the log. Can you share the output of `roslaunch`?

DanielTakeshi, 2023-03-22:
Ah! Sorry about that. I would do that, except I deleted my username on the machine, made a new clean user, and suddenly it went away.

closed 2023-03-22 by DanielTakeshi