IFL-CAMP/easy_handeye #127: Wrong visualisation of point cloud after calibration

TongZhangTHU, 2023-05-14:
Hello,

After I calibrated the RealSense camera extrinsics relative to the Franka Emika Panda using easy_handeye, I visualized the point cloud in the robot frame with rviz /camera/depth/color/points topic (I have published the extrinsics TF), but found that the point cloud is in the wrong position and looks rotated in some way, I suspect that it is caused by the wrong coordinate system. I wonder if you have any suggestions for my problem?

[image]

marcoesposito1988, 2023-06-10:
Hi @TongZhangTHU,

did you visualize the point cloud after starting the system including `publish.launch`, or is still using the dummy position of the camera from the `calibrate.launch` file?

TongZhangTHU, 2023-06-17:
@marcoesposito1988 

I visualize the point cloud from the  `/camera/depth/color/points` topic after starting `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`.

marcoesposito1988, 2023-06-21:
Yes, but after doing the calibration did you restart everything, publishing the hand-eye calibration with [this launch file](https://github.com/IFL-CAMP/easy_handeye/blob/master/easy_handeye/launch/publish.launch), or is the system still running with `calibrate.launch` instead?

After calibration the new transform is not applied automatically, you have to restart everything using a different launch file to use the result of the calibration.

TongZhangTHU, 2023-06-22:
@marcoesposito1988 

The results I presented above were obtained after restarting everything and publishing the hand-eye calibration with [this launch file](https://github.com/IFL-CAMP/easy_handeye/blob/master/easy_handeye/launch/publish.launch), just as you suggested.

marcoesposito1988, 2023-07-09:
It seems that the point cloud is rotated 90 degrees along two axes, so probably a different frame is being used to publish the calibration than what was used during the calibration. Please double check what is passed as frame for the camera

Skeptic9793, 2024-06-03:
> It seems that the point cloud is rotated 90 degrees along two axes, so probably a different frame is being used to publish the calibration than what was used during the calibration. Please double check what is passed as frame for the camera
I'm also experiencing this problem, which seems to be caused by the fact that realsense's optical coordinate system definition is different from the ROS coordinate system definition

Skeptic9793, 2024-06-03:
I'm also experiencing this problem, which seems to be caused by the fact that realsense's optical coordinate system definition is different from the ROS coordinate system definition

Panda12580, 2025-03-15:
> I'm also experiencing this problem, which seems to be caused by the fact that realsense's optical coordinate system definition is different from the ROS coordinate system definition

Have you solved this problem? I ran into this problem too, where I tried to rotate by publishing an intermediate coordinate system and it seemed to work, but it seemed like there was some error in the rotation Angle. Can you tell me your solution?

marcoesposito1988, 2025-11-30:
You may have to use a different frame for the camera optical center and the reference frame of the marker tracking. Please look here for a reference: https://github.com/IFL-CAMP/easy_handeye/blob/master/docs/example_launch/ur5_kinect_calibration.launch#L21