IFL-CAMP/easy_handeye #129: Error when takes sample

ahasfasd, 2023-06-26:
Hi, currently I try to run easy_handeye progam, I follow the procedure in README.md to download the code and deploy the requirements. Everything goes fine until I  plan the first pose and execute, I press Take Sample button in rqt_easy_handeye, and here comes the error:

[ERROR] [1687771262.283389]: Error processing request: "base" passed to lookupTransform argument target_frame does not exist. 
['Traceback (most recent call last):\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 633, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/handsfree/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_server.py", line 88, in take_sample\n    self.sampler.take_sample()\n', '  File "/home/handsfree/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 88, in take_sample\n    transforms = self._get_transforms()\n', '  File "/home/handsfree/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_sampler.py", line 71, in _get_transforms\n    Duration(10))\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 87, in lookup_transform\n    return self.lookup_transform_core(target_frame, source_frame, time)\n', 'LookupException: "base" passed to lookupTransform argument target_frame does not exist. \n']
Traceback (most recent call last):
  File "/home/handsfree/catkin_ws/src/easy_handeye/rqt_easy_handeye/src/rqt_easy_handeye/rqt_easy_handeye.py", line 132, in handle_take_sample
    sample_list = self.client.take_sample()
  File "/home/handsfree/catkin_ws/src/easy_handeye/easy_handeye/src/easy_handeye/handeye_client.py", line 76, in take_sample
    return self.take_sample_proxy().samples
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 523, in call
    responses = transport.receive_once()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_base.py", line 742, in receive_once
    p.read_messages(b, msg_queue, sock) 
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 360, in read_messages
    self._read_ok_byte(b, sock)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 343, in _read_ok_byte
    raise ServiceException("service [%s] responded with an error: %s"%(self.resolved_name, str))
rospy.service.ServiceException: service [/ur5_k4a_handeyecalibration_eye_on_hand/take_sample] responded with an error: error processing request: "base" passed to lookupTransform argument target_frame does not exist.
[ur5_k4a_handeyecalibration_eye_on_hand/namespace_handsfree_10987_2946137096517355517_rqt-4] process has died [pid 11015, exit code -6, cmd /home/handsfree/catkin_ws/src/easy_handeye/rqt_easy_handeye/scripts/rqt_easy_handeye __name:=namespace_handsfree_10987_2946137096517355517_rqt __log:=/home/handsfree/.ros/log/909133dc-13f2-11ee-b835-aced5cd3d4de/ur5_k4a_handeyecalibration_eye_on_hand-namespace_handsfree_10987_2946137096517355517_rqt-4.log].
log file: /home/handsfree/.ros/log/909133dc-13f2-11ee-b835-aced5cd3d4de/ur5_k4a_handeyecalibration_eye_on_hand-namespace_handsfree_10987_2946137096517355517_rqt-4*.log

By the way, my platform is Ubuntu 18.04, Ros melodic, python2.7 and eye_in_hand.
Thanks for your answer in advance

marcoesposito1988, 2023-07-09:
Hi @ahasfasd,

the first line of the log tells you the problem. It looks like you specified `base` as one of the frames of interest for `easy_handeye` (robot base and hand, camera optical center and marker tf frames), but it was not found in the tf tree or it is not connected to the rest of the frames.

Please double check the parameters you are passing to the `easy_handeye` launch file, and that everything is being published in tf.

ahasfasd, 2023-07-12:
Thanks, it helps

closed 2023-08-03 by marcoesposito1988