IFL-CAMP/easy_handeye #132: I am confused because of other online articles introducing easy_hand.

flashmoment, 2024-01-05:
The final result represents the transformation matrix from the robotic arm's base coordinate system to the camera coordinate system, correct?

flashmoment, 2024-01-05:
"I would like to ask, in the 'eye on base' mode, is the result obtained the transformation matrix from the base coordinate system (base) to the camera coordinate system (camera)?"

marcoesposito1988, 2024-01-07:
Hi @flashmoment,

yes, that's correct, although I don't remember the direction of the transformation off the top of my head. You can try inverting the matrix and checking if then it corresponds with what you expect.

You should also pay attention to the tf frames that are being used. In particular when using AR markers with RGBD cameras you may need to publish the result of the marker tracking with respect to the camera's optical frame, but then compute the calibration with respect to the root tf frame of the camera, as done [here](https://github.com/IFL-CAMP/easy_handeye/blob/master/docs/example_launch/ur5_kinect_calibration.launch#L21) for the ROS1 version of easy_handeye