IFL-CAMP/easy_handeye #137: rqt_easy_handeye not starting with roslaunch command

julyfun, 2024-04-17:
os: ubuntu20.04
ros: noetic
date: 24.4.17

When I `roslaunch easy_handeye ur10_cali.launch`(which is newly created), rviz is working correctly, but there is no `rqt_easy_handeye` window (although the `rqt_easy_handeye` node is shown in `rosnode list`). And I don't know how to calibrate without that GUI. Thank you for any suggestion.

This is my `ur10_cali.launch` (the robot is started somewhere else):

```xml
<launch>
    <arg name="namespace_prefix" default="ur10_k4a_handeyecalibration"/>

    <!-- <arg name="ur_robot_driver"
         doc="If true, the new ur_robot_driver will be used; otherwise, the old ur_modern_driver"/>
    <arg name="robot_simulated"
         doc="If true, the demo.launch of ur5_moveit_config will be launched; otherwise, a connection to the real robot will be established"/>
    <arg name="robot_ip" doc="The IP address of the UR5 robot"/> -->

    <arg name="marker_size" doc="Size of the ArUco marker used, in meters"/>
    <arg name="marker_id" doc="The ID of the ArUco marker used"/>

    <!-- start the Kinect -->

    <arg name="rgb_rect" default="1" />
    <arg name="depth_rect" default="1" />
    <arg name="ir_rect" default="1" />
    <arg name="point_cloud" default="1" />
    <arg name="rgb_point_cloud" default="1" />

    <!-- Start the K4A sensor driver -->
    <!-- [j] these should be ok -->
    <group ns="k4a" >
        <include file="$(find azure_kinect_ros_driver)/launch/driver.launch" >
            <arg name="overwrite_robot_description" value="false" />
        </include>

        <!-- Spawn a nodelet manager -->
        <node pkg="nodelet" type="nodelet" name="manager" args="manager" output="screen">
            <param name="num_worker_threads" value="16" />
        </node>

        <!-- Spawn an image_proc/rectify nodelet to rectify the RGB image -->
        <node if="$(arg rgb_rect)"
              pkg="nodelet" type="nodelet" name="rectify_rgb"
              args="load image_proc/rectify manager --no-bond"
              respawn="true">
            <remap from="image_mono"  to="rgb/image_raw" />
            <remap from="image_rect"  to="rgb/image_rect_color" />
        </node>

        <!-- Spawn an image_proc/rectify nodelet to rectify the depth image -->
        <node if="$(arg depth_rect)"
              pkg="nodelet" type="nodelet" name="rectify_depth"
              args="load image_proc/rectify manager --no-bond"
              respawn="true">
            <remap from="image_mono"  to="depth/image_raw" />
            <remap from="image_rect"  to="depth/image_rect" />

            <param name="interpolation" value="0" />
        </node>

        <!-- Spawn an image_proc/rectify nodelet to rectify the IR image -->
        <node if="$(arg ir_rect)"
              pkg="nodelet" type="nodelet" name="rectify_ir"
              args="load image_proc/rectify manager --no-bond"
              respawn="true">
            <remap from="image_mono"  to="ir/image_raw" />
            <remap from="image_rect"  to="ir/image_rect" />
        </node>

        <group if="$(arg point_cloud)">
            <!-- Spawn a depth_image_proc/point_cloud_xyz nodelet to convert the
                depth image into a point cloud -->
            <node unless="$(arg rgb_point_cloud)"
                  pkg="nodelet" type="nodelet" name="point_cloud_xyz"
                  args="load depth_image_proc/point_cloud_xyz manager --no-bond"
                  respawn="true">
                <remap from="image_rect"  to="depth/image_rect" />
            </node>

            <group if="$(arg rgb_point_cloud)">
                <!-- Spawn a depth_image_proc/register nodelet to transform the
                    depth image into the color camera co-ordinate space -->
                <node pkg="nodelet" type="nodelet" name="depth_register"
                      args="load depth_image_proc/register manager --no-bond"
                      respawn="true">
                </node>

                <!-- Spawn a depth_image_proc/point_cloud_xyzrgb nodelet to convert the
                    depth_registered and color images image into a colorized point cloud -->
                <node pkg="nodelet" type="nodelet" name="point_cloud_xyzrgb"
                      args="load depth_image_proc/point_cloud_xyzrgb manager --no-bond"
                      respawn="true">
                </node>
            </group>
        </group>

    </group>

    <!-- start easy_aruco to track the example board -->
    <include file="$(find easy_aruco)/launch/track_charuco_board.launch">
        <arg name="camera_namespace" value="/k4a/rgb"/>
        <arg name="dictionary" value="DICT_6X6_250"/>
        <arg name="square_number_x" value="3"/>
        <arg name="square_number_y" value="3"/>
        <arg name="square_size" value="0.024"/>
        <arg name="marker_size" value="0.016"/>
        <arg name="reference_frame" value="camera_base"/>
        <arg name="camera_frame" value="rgb_camera_link"/> <!-- [j] yes we have that frame -->
    </include>

    <!-- start the robot -->
    <!-- <group unless="$(arg robot_simulated)">
        <include if="$(arg ur_robot_driver)" file="$(find ur_robot_driver)/launch/ur5_bringup.launch">
            <arg name="limited" value="true"/>
            <arg name="robot_ip" value="192.168.0.21"/>
        </include>
        <include unless="$(arg ur_robot_driver)" file="$(find ur_bringup)/launch/ur5_bringup.launch">
            <arg name="limited" value="true"/>
            <arg name="robot_ip" value="192.168.0.21"/>
        </include>
        <include file="$(find ur5_moveit_config)/launch/ur5_moveit_planning_execution.launch">
            <arg name="limited" value="true"/>
        </include>
    </group>
    <include if="$(arg robot_simulated)" file="$(find ur5_moveit_config)/launch/demo.launch">
        <arg name="limited" value="true" />
    </include> -->
    <!-- we don't start RViz, we have our own -->

    <!-- start easy_handeye -->
    <include file="$(find easy_handeye)/launch/calibrate.launch">
        <arg name="namespace_prefix" value="$(arg namespace_prefix)"/>
        <arg name="eye_on_hand" value="false"/>
        <!-- <arg name="start_sampling_gui" value="true"/> -->

        <arg name="tracking_base_frame" value="camera_base"/>
        <arg name="tracking_marker_frame" value="board"/>
        <arg name="robot_base_frame" value="ra_base_link"/>  <!-- before is base_link -->
        <arg name="robot_effector_frame" value="ra_wrist_3_link"/>

        <arg name="freehand_robot_movement" value="false"/>
        <arg name="robot_velocity_scaling" value="0.5"/>
        <arg name="robot_acceleration_scaling" value="0.2"/>
    </include>

</launch>
```

my `rosnode list`:

```
/arm_trajectory_controller_spawner
/calibrate_sr_edc
/conditional_delayed_rostool_action_multi
/conditional_delayed_rostool_diagnostic_aggregator
/conditional_delayed_rostool_movegroup_multi
/conditional_delayed_rostool_robot_state_publisher
/control_box/bag_rotate_nuc_control_7073_7770324356114935517
/control_box/core_dump_limit_nuc_control_7073_3367785699137591959
/control_box/journalctl_pub_nuc_control_7073_2390878166595180627
/control_box/record
/controller_stopper
/diagnostic_aggregator
/dummy_handeye
/easy_aruco_node
/error_reporter
/joint_0_pub
/joint_state_controller_spawner
/k4a/azure_kinect_ros_driver
/k4a/depth_register
/k4a/joint_state_publisher_azure
/k4a/manager
/k4a/point_cloud_xyzrgb
/k4a/rectify_depth
/k4a/rectify_ir
/k4a/rectify_rgb
/k4a/robot_state_publisher_azure
/mongo_wrapper_ros_nuc_control_7642_404583385217236712
/move_group
/moveit_python_wrappers_1713334404607131791
/moveit_python_wrappers_1713335908435908361
/moveit_warehouse_services
/robot_state_publisher
/rosout
/rqt_gui_cpp_node_20861
/rqt_gui_cpp_node_24272
/rviz_server_23634_6718413383284208217
/sr_hand_robot
/sr_rh_tactile_sensor_controller_spawner
/sr_ur_arm_unlock
/teach_mode_node
/ur10_k4a_handeyecalibration_eye_on_base/calibration_mover
/ur10_k4a_handeyecalibration_eye_on_base/easy_handeye_calibration_server
/ur10_k4a_handeyecalibration_eye_on_base/namespace_server_15282_2785989410345704368_rqt
/ur10_k4a_handeyecalibration_eye_on_base/namespace_server_23634_7169102761148973672_rqt_sampling_gui # the sampling gui node is renamed to this
```

julyfun, 2024-04-17:
from this video [on bilibili](https://www.bilibili.com/video/BV1Gu411s7cd) we can see that rqt_easy_handeye window is shown when the command is executed.

Fanqyu, 2024-04-17:
Hi, cannot see the plugin, too.
[image]
Have you solved this problem?

julyfun, 2024-04-17:
> Hi, cannot see the plugin, too. [...]

not yet

julyfun, 2024-04-23:
> If it's convenient, we can add the WeChat account of each other, and discuss the problem together.

ok you may email me your wechat account 😊

ChuckAlex, 2024-08-05:
hi， Have you solved this problem?

Gritsua, 2024-09-18:
hi， Have you solved this problem,and can i email my wechat to you,discussing the problem?

tyb197, 2024-12-23:
> r

Hello, I also had this problem, there are two rviz whidows displayed but no rqt_easy_handeye window. I had checked the move_group and the default is arm, it seems correct. If it is convinent, can I have your wechat to consult you about
 this question? Thanks a lot~

tyb197, 2024-12-23:
我的launch文件如下：
`<?xml version="1.0" ?>
<launch>
    <arg name="namespace_prefix" default="kinova_kinect_handeyecalibration" />
 
    <arg name="marker_size" value="0.1" />
    <arg name="marker_id" value="325" />
 
    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/kinect2/hd/camera_info" />
        <remap from="/image" to="/kinect2/hd/image_color_rect" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="kinect2_rgb_optical_frame"/>
        <param name="camera_frame"       value="kinect2_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
 
    </node>
 
    
    <include file="$(find kinova_bringup)/launch/kinova_robot.launch">
    </include>
    <include file="$(find j2n6s300_moveit_config)/launch/j2n6s300_demo.launch">
    </include>
    <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
    </node>
    
    
    <include file="$(find kinect2_bridge)/launch/kinect2_bridge.launch">
    </include>
    
 
    
    <include file="$(find easy_handeye)/launch/calibrate.launch" >
        <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
        <arg name="eye_on_hand" value="false" />
 
        <arg name="tracking_base_frame" value="kinect_rgb_optical_frame" />
        <arg name="tracking_marker_frame" value="camera_marker" />
        <arg name="robot_base_frame" value="j2n6s300_link_base" />
        <arg name="robot_effector_frame" value="j2n6s300_end_effector" />
        
        
        <arg name="freehand_robot_movement" value="false" />
        <arg name="robot_velocity_scaling" value="0.5" />
        <arg name="robot_acceleration_scaling" value="0.2" />
    </include>
 
</launch>
`
calibrate文件中move_group如下：
`<arg name="move_group" default="arm"  doc="the name of move_group for the automatic robot motion with MoveIt!" />`

Benxiaogu, 2024-12-25:
> 我的launch文件如下： `
> 
> ```
> <arg name="marker_size" value="0.1" />
> <arg name="marker_id" value="325" />
> 
> 
> <node name="aruco_tracker" pkg="aruco_ros" type="single">
>     <remap from="/camera_info" to="/kinect2/hd/camera_info" />
>     <remap from="/image" to="/kinect2/hd/image_color_rect" />
>     <param name="image_is_rectified" value="true"/>
>     <param name="marker_size"        value="$(arg marker_size)"/>
>     <param name="marker_id"          value="$(arg marker_id)"/>
>     <param name="reference_frame"    value="kinect2_rgb_optical_frame"/>
>     <param name="camera_frame"       value="kinect2_rgb_optical_frame"/>
>     <param name="marker_frame"       value="camera_marker" />
> 
> </node>
> 
> 
> <include file="$(find kinova_bringup)/launch/kinova_robot.launch">
> </include>
> <include file="$(find j2n6s300_moveit_config)/launch/j2n6s300_demo.launch">
> </include>
> <node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher">
> </node>
> 
> 
> <include file="$(find kinect2_bridge)/launch/kinect2_bridge.launch">
> </include>
> 
> 
> 
> <include file="$(find easy_handeye)/launch/calibrate.launch" >
>     <arg name="namespace_prefix" value="$(arg namespace_prefix)" />
>     <arg name="eye_on_hand" value="false" />
> 
>     <arg name="tracking_base_frame" value="kinect_rgb_optical_frame" />
>     <arg name="tracking_marker_frame" value="camera_marker" />
>     <arg name="robot_base_frame" value="j2n6s300_link_base" />
>     <arg name="robot_effector_frame" value="j2n6s300_end_effector" />
>     
>     
>     <arg name="freehand_robot_movement" value="false" />
>     <arg name="robot_velocity_scaling" value="0.5" />
>     <arg name="robot_acceleration_scaling" value="0.2" />
> </include>
> ```
> 
> ` calibrate文件中move_group如下：
> `<arg name="move_group" default="arm" doc="the name of move_group for the automatic robot motion with MoveIt!" /> `

hello, I am having the same problem as yours. Have you solved this problem?

Gritsua, 2024-12-25:
您好！

谢谢，非常感谢你给我提供的launch文件！我们遇到过这个问题，后来我们检查了生成的tf tree,发现了一些问题，改动了tf tree之后，我们就能正常标定了，再次感谢你的帮助！！如果有幸能添加您的微信，那就太好啦！我的微信号是：Gritsua

祝好！
sua

------------------&nbsp;原始邮件&nbsp;------------------
发件人:                                                                                                                        "IFL-CAMP/easy_handeye"                                                                                    ***@***.***&gt;;
发送时间:&nbsp;2024年12月25日(星期三) 下午3:53
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [IFL-CAMP/easy_handeye] Load rqt_easy_handeye plugin failed (Issue #137)

  
我的launch文件如下： `
 <arg name="marker_size" value="0.1" /&gt; <arg name="marker_id" value="325" /&gt;

LiperLee, 2025-05-16:
Hi, Has the problem been solved? There is no problem with move_group manipulator either

marcoesposito1988, 2025-11-30:
I just tried to reproduce the error on 20.04 with an up-to-date ROS noetic. The workspace just contains easy-handeye, the demo package, and ros-industrial/universal_robot at current noetic-devel. Everything runs as expected.

Please check that you have all dependencies and that your installation doesn't deviate from the standard.