IFL-CAMP/easy_handeye #141: calibration

Panda12580, 2024-05-22:
Hey, sorry to bother you. I'm having some issues with hand-eye calibration and I could use your help. First, I'm using ArUco to detect QR codes on my hand, and I need to calibrate the camera intrinsic parameters (RealSense 435) before performing the calibration. How do I do this? Or if I already have the intrinsic matrix and distortion parameters, how should I use them? Second, I've noticed that the results of my calibration (hand outside the camera) seem reasonable in terms of the camera's xyz relative to the base coordinates, but the xyzw seems to have a large deviation. I've checked my reference coordinate system, but I can't seem to solve this issue. I hope you can help me. Thanks.

SakethTecholution, 2024-10-22:
U  don't need to callibrated intrinsics for a depth camera since it auto callibrated and u can get it 
rs-enunerate-devices -c

marcoesposito1988, 2025-11-30:
Hi @Panda12580,

if you are using an RGBD camera, they usually are indeed factory-calibrated as suggested by SakethTecholution.

For the second problem, you may have a confusion in your tf frames. Please check the FAQ.

closed 2025-11-30 by marcoesposito1988